BenjaminPoilve/minichord #6: The MPR121 capacitive touch sensor controller is discontinued

sampsonlau, 2024-12-02:
The MPR121 capacitive touch sensor controller by NXP was discontinued in 2019, and most distributors no longer carry it. Recommend replacing with a controller that is actively manufactured and supported. I am not aware of any pin-compatible replacements, so a PCB design update may be necessary.

BenjaminPoilve, 2024-12-02:
Hi!
Indeed it was dropped from production by NXP. This is the reason why the initial version of the minichord used something from the AT42QT line, but the 16ms update cycle was really too slow (compared to 1ms for this one). 
The good news is that, from my understanding, the license has been sold to Resurgent Semiconductors in 2015. 
[Arrows shows 542598 parts available](https://www.arrow.com/en/products/mpr121qr2/resurgent-semiconductor-llc), so we might be ok for a while. 
If anybody knows alternative parts though, I'm all hears!

sampsonlau, 2024-12-02:
Ah, the AT42QT was actually the closest thing I found that's in production. Good to know Arrow still has stock, but I'm not a fan of the 1000-qty minimum order increment. 😅

BenjaminPoilve, 2024-12-02:
We can make a grouped order, I know this chip is quite popular with DIYers (and rightly so, it is great) :)