BenjaminPoilve/minichord #7: Extraneous openings on solder paste stencil between R3 and R7

sampsonlau, 2024-12-09:
In the space between R3 and R7, there appears to be an extra pair of openings on the solder paste stencil that does not align with any solder pads. In the attached image, they are the solder pads highlighted in yellow.
[image]

BenjaminPoilve, 2024-12-09:
Well that's a careful review you did to catch that :)
Yes indeed it should not be on paste. This does not cause any issues, because that footprint is simply used to bridge ground and power ground on a single point. 
I'll fix it shortly!

BenjaminPoilve, 2024-12-15:
Will be removed in the next commit

closed 2024-12-15 by BenjaminPoilve