BenjaminPoilve/minichord #9: SMD LED

KeenVox, 2025-01-20:
Hi there,

I have trouble finding the correct dual smd led on mouser, digikey etc, 

Could you provide a part number, link or description of that part? :-)

KeenVox, 2025-01-20:
Hi Ben,
thx!
I think I found the equivalent on mouser:

https://www.mouser.de/ProductDetail/Kingbright/APFA3010R2G2C-C6-AMT?qs=amGC7iS6iy%2F4sATRKhbw8A%3D%3D

KeenVox, 2025-01-20:
Ah, nevermind. The pinout doesn't match.

KeenVox, 2025-02-08:
So I'm here again. I've got some problems with the status leds. I hooked some up like in the schematic. But the green LED is always on. The red LED is on, when I plug in the USB power but very dim.

Anything I'm missing here?

BenjaminPoilve, 2025-02-08:
Hello again, 
Huum I see two options: 
- Do you have the battery plugged in? If not the controller will blink both led, which can gives strange results
- Those led are a pain to solder. It is very easy to create short circuit (event making sure it is the right orientation and it is not upside down is no small task). I've had situations where I had to give up reworking a led because I just did not managed to do it with a solder (and the board could not be reflow).

KeenVox, 2025-02-08:
Hey, I found the problem. My msitake. I bought a MCP73832 instead of a MCP73831. The former one has a open drain on the STAT pin. I assume that's the problem. Will see, when the new 831 arrives.
As for the LED In used some normal 3mm LEDs and soldered those to the resistors as I was not able to buy the needed Bi-Color LED on any other source. :)

BenjaminPoilve, 2025-04-20:
BOM has been updated, so I close :)

closed 2025-04-20 by BenjaminPoilve