BenjaminPoilve/minichord #10: Sharp button crashes system

KeenVox, 2025-02-03:
Hi, 

I've got a problem with the minichiord. Everything works fine. The touch controls, buttons and pots. 

But as soon as I press the sharp button together with a button of the first row the system seems to crash. The LED turns off and the system stays of as long as I hold the sharp and any other button of the first row together. When I let go the system resets and works again on the first preset.

Using the sharp button with the second or third row works fine.

What could be the cause?

BenjaminPoilve, 2025-02-03:
Hi, 
That sure looks like a brownout that would be caused by a shorting of the 3.3V line to the ground. 
Inspect the PCB for any soldering defect, and most importantly, reflect on what is below the PCB. It happened to me that pushing the button on the end of the PCB was enough to create flex in the PCB. The flex can make the leads of the pushbutton dig into the battery and create shorts. 
Two solutions: 
- Use a thick double sided tape as recommanded in the instruction for the old version 
- For the new version, the battery has been moved around to avoid that issue altogether. 
Let me know how that goes. 
Regards, 
Ben

KeenVox, 2025-02-03:
Hi Ben,

I don't have the the battery connected. I get power over USB directly from the Teensy, The pcb is outside of the case.
My soldering looks fine, I checked for continuity and can't find any obvious shorts. Measuring the sharp button in operation, I can see it gets 3.3V and goes to 0V when pressed. 
It only resets when sharp and any button from the read matrix 1 are pushed together. Checking all buttons neither one goes to GND when pressed. 
On the 74HC595 I also get no false ground connection. Only Pin 8 and 13 are connected to ground. :(

BenjaminPoilve, 2025-02-03:
Hi again, 
Then we need to get into diagnostics mode :)
First it would be nice to confirm the brownout. The 3.3V regulator is on pin 31, ground on pin 1 of the Teensy. Maybe use a scope to confirm that the voltage is dropping when pressing both buttons? 

Then, just a reminder of how this section of the minichord works. All the buttons of the first row (and the sharp) are connected to the output of the multiplexer. The other side is held high by a pullup resistor. The outputs of the multiplexer are held high, and get low one by one to detect which of the switches is pressed. Now this is a little unorthodox as you might notice there is no matrix diode. But due to the button patterns that are expected, there is no problem to detect what is happening. 

Once thing that strikes me as odd is the fact that the sharp goes to zero when pressed. Technically it does not go to zero, it goes to the state of the output of the multiplexer that is zero 1/8th of the time. 

In addition do you have an issue when you press two buttons from the first line at the same time? Does it cause the same brownout?

BenjaminPoilve, 2025-02-03:
(I'll check tonight to make sure I can't reproduce the issue, but it would be strange to detect it for the first time now when more than a few minichords are out there)

BenjaminPoilve, 2025-02-03:
To add: where did you buy your 74HC595 ? My design (maybe not so wisely) relies on the fact that when two state outputs are tied together, the current is limited. Maybe your chip has a different behavior? I might need to think about adding diodes on the 74HC595 pins to avoid that problem.

KeenVox, 2025-02-03:
Hey,

I checked the voltages on Pin31 on my scope. I've a little pulsing in it but minimum is 3,1V and max 3,3V. When sharp and another button a pressed it goes down to minimum  2.9V.
I rechecked the sharp button with my multimeter and it indeed goes down from 3.3V to 0V when pressed.
I checked the 3.3.V and 5V Pin also. Nothing changes there.
When two buttons are pressed the led on the teensy and the big status led on the board go out. In my headphones I can hear a kind of pulsing sound.
It only happens when the sharp button and one button of the read matrix (1-7) are pressed toghether. All other button combination don't show that behaviour.  

I attached two pictures from my board. The smd parts are hand soldered but all values are correct and continuity is where it should be. :) No shorty to be found anywhere.
HC
My 74HC595D is from NXP. I only checked the pinout against an TI 74HC595 and at least those a lining up correct.

[image]
[image]

BenjaminPoilve, 2025-02-03:
I'll give it more thoughts this evening, but I have to say, that is some impressive soldering! Your board looks cleaner than mine :)

Since it only happens with the sharp when all the button of the line and the sharp are similarly wired, it seems to indicate it is something else that a different 74HC595 behavior..

KeenVox, 2025-02-03:
Thanks, I have some projects on my belt ;)

I went the mile and desoldered all the buttons from the first row an even desoldered the teensy aaannd cut the led trace juuust to be sure. 
I just used tweezers for the chord buttons but it still crashes when I press the sharp button. :(

BenjaminPoilve, 2025-02-03:
The trouble seems to be that the sharp is connected to ground. It should not be. 
Here is the pinout of the chip 

[image]

Could there be a bridge between the two right bottom pin? Or below the chip itself?

KeenVox, 2025-02-03:
I checked the pins, at least with power off there is no false GND connection. The switch itself has no continuity to GND also, pressed nor unpressed.

I think I have to go the exclusion process and will swap the 595. Sadly of all the parts in my stash I only had one piece. So will take a day or to two till I can try that. Hope it's simply the IC. Can't think of another error. I checked all the traces to the teensy and even desolderen all the row 1 buttons. 

The only thing that is not connected is the battery as I'm waiting for the right one to be delivered, but I can't think of that as the problem...

BenjaminPoilve, 2025-02-04:
Well I'm sorry about the experience, it is not great :(
Reflecting a bit more about it, there is something special about that sharp button. When looping over the buttons, the last one in the loop is the sharp, so it stays to the ground until the next button poll. Why that would cause a short in your situation, I'm not sure, but to see if that could be a cause, I'll modify the firmware to put the pin back up after polling. Try it and let me know if that fixes your situation.

KeenVox, 2025-02-04:
Thx, will try, when I'm back home tonight. :)

We'll get it too work, I'm sure.

KeenVox, 2025-02-04:
Hey, good news,

everything is now working like it should. Thank you!

BenjaminPoilve, 2025-02-05:
That's a good news! For my understanding, that is with the firmware fix? 
If that is the case, I still have to do a PCB run with added diodes to test. Running outside of specs is never great!

BenjaminPoilve, 2025-02-05:
Thanks! 

And in particular, thanks for building a minichord and raising that issue. That'll help me better the design :)

Hope you'll have fun, and don't hesitate if you got features you want to be added!

KeenVox, 2025-02-05:
Thank you for designing this.

I know what a hassle it can be to design pcbs :-D

Sadly I fried my Teensy yesterday after attaching the battery. Will have to
sort that out. Hopefully this won't be such a hassle as the sharp button ;)

BenjaminPoilve, 2025-02-05:
Oh. Did you plugged the battery in reverse? Happened to me once, and fried both the Teensy and the battery controller. Unfortunately, I did not find an easy way to defend against this, as most simple solutions have significant drawbacks. I tried to make it clear on the silkscreen, but if you have advice to apply poka yoke here, I'm all ears!

KeenVox, 2025-02-05:
No, I tripple checked. Don't really know what happened. Maybe a short I
didn't see. I rechecked everything but seems to be fine.

Only thing I noticed was that I have a v0.3 pcb and forgot to clip the
jacks pin and solder a 10K to the enable. Don't know if that could be the
cause?

For the protection, I think that will get complicated fast. Normally I
would simply use a diode in series or a mosfet but here we have the
charging circuit as the problem.
I think that would need a bit more effort...

BenjaminPoilve, 2025-02-06:
It could be the cause, as some switched jack are "make before break switch", which would short the GND and Battery power when plugging in. This is the reason for the later revision, that does not connect GND, but pulls the enable pin of the power chip with a resistor. Indeed earlier PCB need that modification. 

Thanks to your feedback I made an update to the PCB to add diodes to the multiplexer. Should be more sturdy :)

BenjaminPoilve, 2025-02-06:
Thanks :) 

Yep, I have added it to the repo as well, it is the "Keymaster" PCB. When I have the time I'll make a Kicad template so people can design their own custom touchplate!

KeenVox, 2025-02-06:
Great. Will look into it. :)

BenjaminPoilve, 2025-02-07:
With the latest commit, I've added better gerber (with silkscreen) for the widest touchzone, and an easy to modify kicad template for custom touchzones.

KeenVox, 2025-02-07:
Perfekt! Thx!

BenjaminPoilve, 2025-04-11:
I'll close this issue since added diodes for the next version make the initial crash impossible, even with original code.

closed 2025-04-11 by BenjaminPoilve

KeenVox, 2025-04-11:
Good to know. Will my pcb without the diodes be still compatible if you
update the firmware any further? Or will you remove the fix from it?

BenjaminPoilve, 2025-04-11:
There is no reason to roll back the change, so you'll stay compatible :)