BenjaminPoilve/minichord #12: Change strum zone size

BenjaminPoilve, 2025-02-09:
Following feedback on the fact that the current shape makes it easy to scrape the edge of the touch plate, there is a need to widen the top and bottom clearance between pads and edge of the touchplate

YonkRaccoon, 2025-04-10:
To diverge from this topic, is there a digital parameter to change the "strum range" of a synth preset? For example, making it so strumming across the sensors only span 4, or even 2 notes? It would be useful for recording specific kinds of parts to my songs.

BenjaminPoilve, 2025-04-11:
To update on the initial issue, the version that will be produced by Seeed will have a slightly larger touch zone size. I'll upload the updated files later. 
For your second question, the notes associated to each of the strum zone is indeed user-settable, per preset. It is defined [here](https://github.com/BenjaminPoilve/minichord/blob/36667c45511398d1447d1aac58f1e4aa2e58a66f/firmware/src/main.cpp#L142-L150). 
If you look at the [minicontrol interface](https://minichord.com/minicontrol/), parameter 40, you can see the related control.

YonkRaccoon, 2025-04-11:
Thank you for linking the minicontrol interface! I had no idea we had access to it already before public release. You are truly going open source, which is something I don't get to dabble with much yet in my life.

**As for changing the PCB, where can someone like me buy the plate which you tap the dots for notes?** I saw it in an early demo video in the past and didn't know if it was sold ready-made somewhere.

That's my last question in this topic! <3

terminalwaltz, 2025-06-21:
@YonkRaccoon I'm also interested in a keymaster strum plate: really want to see how it performs with Barry Harris mode. I've used JLCPCB as a service in the past: I'm pretty sure they have at least a 5 PCB minimum, and I have no use for 5 touch plates.

If you'd like, I'll let you know when I've got a batch and I can send you one. I remember their turn around time was pretty good in 2021, hopefully that hasn't changed.

terminalwaltz, 2025-06-21:
@YonkRaccoon Order placed! Let this post be a placeholder, I'll update when I get the boards and then we'll have an average time of the complete build/shipping for anyone in the future who wants to order (to the eastern US, at least). Cool time to be a gadget nerd :D

terminalwaltz, 2025-06-21:
No, I ordered 5 PCBs for the keymaster strum plate because the company that is going to print them (JLCPCB) requires a minimum of 5 to fill an order. So they're gonna send me 5, and I can't imagine a scenario in which I would need all 5. This is such a niche community, and I saw that you were also interested in the keymaster plate, so I figured I'd offer to send you one. 

The total cost for printing the 5 PCBs was a whopping $6.05, after tax and shipping. And I'll solder some headers on there for you, if you'd like, easy stuff. I just think it would be neat to make a total stranger happy in this very odd, very specific way 😄 

For that matter, I'll have another 3 left over, but I'm not trying to start a riot. But if anyone else wants one, I'm happy to send them out!

YonkRaccoon, 2025-06-22:
Aw, dude that rules. I relate cause I would try to do the same if I had the resources currently. I feel very lucky to be vocal right now about my passions

-------- Original Message --------

BenjaminPoilve, 2025-06-22:
Hi there! Just in case anyone has doubt, this is fully OK to manufacture, share, or sell alternative touchplates. You have my express permission to do so :) 
Thanks for helping out another user @terminalwaltz!

YonkRaccoon, 2025-06-29:
> For your second question, the notes associated to each of the strum zone is indeed user-settable, per preset. It is defined [...]

Now that I have my minichord, I'm making something as close as possible to an acoustic guitar (ideally wanting to make something as satisfying as the electronic guitar in the soundtrack from the Stray game.)

My goal is to make the harp hit only 6 or 8 notes throughout its entire contact. Do I need to learn how to edit this code in order to reduce the harp strum range? So far, this parameter is useful but only changes what notes the harp will include on its journey.

YonkRaccoon, 2025-06-29:
> [@YonkRaccoon](https://github.com/YonkRaccoon) I'm also interested in a keymaster strum plate: really want to see how it performs with Barry Harris mode. I've used JLCPCB as a service in the past: [...]

Just to let you know, I finaly recieved my minichord and am working on my first niche presets to make very interesting things, with where the software is at so far. You can reach me here about the PCB shipment stuff or the contacts on the bottom of my WIP github website!

BenjaminPoilve, 2025-06-29:
>My goal is to make the harp hit only 6 or 8 notes throughout its entire contact. Do I need to learn how to edit this code in order to reduce the harp strum range? So far, this parameter is useful but only changes what notes the harp will include on its journey.

We would need to discuss how we'd like to do it. It would mean that some contact would not fire. So the question would be: which ones? An alternative would also be to do a touch zone with less zones, and that would need no code modifications. 

>[A Stray Guitar.txt](https://github.com/user-attachments/files/20968147/A.Stray.Guitar.txt)
Nice! Would you like it on minishop as well?

YonkRaccoon, 2025-06-29:
This GitHub is getting me inspired to actually record video and demonstrate different presets and hardware concepts!

I'll try to form a good reply to some things soon, through video or text.

I will reiterate that I'm coming purely from a musician/player side when it comes to Minichord features. I don't know the technological or programming side of anything technical, but I have a lot of ideas based on expressive live performance & experimenting with multiple instruments.

terminalwaltz, 2025-07-06:
@YonkRaccoon The keymaster plates arrived yesterday! I'll be back home this afternoon to solder up and try them out. Shoot me an email with your details and I'll get you one mailed out this week: 

terminalwaltz@gmail.com

terminalwaltz, 2025-10-18:
> @terminalwaltz 
> I'm guessing you've found homes for the extra keymaster plates in your order by now, but figured I'd check? Ha

Yes, unfortunately, the extra keymasters have all found homes scattered around the world.

To anyone considering placing a manufacturing order for their own keymaster: it's very easy, don't be intimidated, check out Ben's new resources, and don't hesitate to ask questions! :)