BenjaminPoilve/minichord #15: Feature (Software:) Custom Startup Jingle

YonkRaccoon, 2025-04-10:
I'm a person who plays around a lot and was wondering if it was possible to create a short MIDI segment and put it into the minichord's brain so when you turn it on, it would play. This would either happen with the preset it turns on with or a specified preset. As I assume, this is such a small amount of data, so I was wondering if it was possible. Thanks for reading!

BenjaminPoilve, 2025-04-11:
Not impossible at all! We are still far from reaching the limit of what the Teensy can do. 
When you say midi segment, you still would like for it to play on the audio output? Or only on the MIDI channels? 
I think the need needs to be further explained: would it be a simple indicator that it turned on, or would you like to use it for musical purposes?

YonkRaccoon, 2025-04-11:
Yes-- to play a short, user-defined tune, on the audio output, when the power is turned on, so it feels an old Mac computer, haha. It would be an indicator, but I can see it being used to also play the first chord or notes of a song. If the minichord has an option to remember which preset synth you were last on, it could simply play the MIDI using that preset synth. But it would be ideal if you could assign which of the 12 preset synths play the MIDI.

If it had more "brain space" in its storage than I assumed, it would be truly ideal for each preset to have **its own** assignable measure of MIDI notes.

I mentioned "MIDI" as it is the only way I know how to "store audio" on the minichord, but maybe use arpeggio patterns are something other than MIDI and that format is also an option? I only know these topics on the consumer level, as I mostly compose with acoustic instruments.

Of course, I already preordered the minichord, so this feature doesn't decide whether or not I pay for it, but it would make me very happy continuing to use it!