BenjaminPoilve/minichord #19: first batch assembly issue (keycaps order and vbat connection) - fix procedure

BenjaminPoilve, 2025-05-27:
As minichords started to ship last week, this should be a happy time, **but unfortunately some issues with the initial batch has been identified which justify this thread** (and a very big thank you to those who did, especially Pierre from [nuancescontroller](https://nuancescontroller.fr/)). **It will serve as a guide to solve the issue and as a discussion zone for technical help.** 

It seems that there has been mistakes in the assembly and procurement that affected those first unit dispatched. I cannot confirm yet if all units are affected, but I’m going to go with the assumption that it is the case. **Of course Seeed Studio immediately stopped shipping until the situation is resolved**. None of those issue were present in the engineering samples I reviewed. It is true though that no mistake happen in a vacuum, and I should have made sure that the process set up was thorough enough to catch those before shipping. “Trust but verify” proves once again a useful proverb.

### Scope of the issue 

The identified errors are the following:

* **The order of the keycaps is wrong**. While current documentation and assembly document provided indicates that the order is F/C/G/D/A/E/B from left to right, the unit shipped use the older key scheme.

* **The bridge between the USB power supply and regulated internal power was not cut**. As a result, the minichord starts when plugged in to the USB port, which is not the expected behaviour and might impact charge time.

There is a final issue related to the size of the provided USB cable, which is too big to be used. On that, Seeed Studio confirmed that they will propose to ship a replacement. 

### Confirm that your unit is affected 

#### Keycap order 
If your unit keycaps read, from left to right G/B/A/F/D/E/C, then the order is wrong. The right order is F/C/G/D/A/E/B.

[image]

#### Vbat bridging 
To test if the unit is affected,  connect the minichord to a power source without plugging in a headphone jack. If both the charging led and the main minichord led turn on, then your unit if affected. 

[image]

### Procedure to fix 

**The procedure to fix both those issues is quite easy**. It takes about 10 minutes to do and require two things: 

* An electronic (small) phillips screwdriver
* An X-acto/hobby knife/small cutter knife

I've recorded the steps to follow [in the video here](https://www.youtube.com/watch?v=eTFiQu6iImM), but please read the instruction carefully to ensure a proper fix. 

#### 1 - Open the minichord

Turn over the minichord and locate the 6 screw holes. There is a seventh hole in the middle: this is the reset button, there is no screw here. Using an appropriately sized screwdriver, unscrew those screws. If your screwdriver cams out, then you have the wrong size. 

[image]

You may need to tap gently the minichord for the screws to drop. 

Once that is done, remove the touch zone. To do that use your nails to lift it, alternate from side to side to keep it even. 

[image]

You can then remove the front enclosure easily. 

#### 2 - Reorder the keycaps 

For each keycap of the first line, use your nails to lift it up while using your thumbs to ensure it stays vertical. Wiggle it a bit until it pops out.  Start from the side to make it easier. 

[image]

Once that is done, you can put them back in the right order (F/C/G/D/A/E/B). 

#### 3 - Cut the Vbat bridge 

This step can feel a little intimidating but it is really easy. Just make sur you have a hobby knife in good shape to facilitate the cutting. The Teensy is designed to allow this cutting, so there is a big pad just to that end. 

First locate the pad using the reference picture below: 

[image]

The red line shows where it needs to be cut. 

Place the hobby knife there and apply pressure while moving a bit to allow cutting. As soon as you feel it is enough, you can plug the usb cable back to power it on. Then there is two option : 
* If the main led turns on in addition to the charging led, then you need to cut more as there is still contact 
* If not, then you are good to go 
This test can be seen in the video. 

#### 4 - Reassemble the minichord

Put back the front enclosure and screw the screws back on. Do not overtighten them: once you feel a resistance, that is enough. 
Put back the touchplate, making sure the header pin are correctly aligned.

That's it! 

### Closing words

While this is not exactly the first experience I hoped for minichord user, I hope this guide will make is as painless as possible. Thanks for bearing with us until all issues are ironed out (I might have to look into designing some “I’m sorry we messed up” special edition alternative touchplates). If you have any question, please use the thread as it might be useful to other users!

BenjaminPoilve, 2025-05-27:
This is not standard behaviour (and departure from my assembly instruction), so I had to work out last night what could be the effect. 
 
- For sure there is the issue of the minichord being powered on while charging: it means that the charging will be slower, and it could impede the charge manager, keeping it from detecting the battery as full (because in practice it would never be). That could on the long term cause battery life issue if you leave it charging continuously. 
- I've identified at least one situation where this blocks the charging: if the battery is totally empty, then the battery manager chip will wrongly read the battery as full (due to the voltage of the USB feeding back) so block the start of the charge. Removing then re-plugging the USB solves it. 

I can't confirm that there won't be any more ill-effect. So my advice would be to do the fix, it is not complicated, the teensy is designed to do it, so if you are opening it up to switch the keycaps, might as well do it.

AlexVernanchico, 2025-05-27:
> I can't confirm that there won't be any more ill-effect. So my advice would be to do the fix, it is not complicated, the teensy is designed to do it, so if you are opening it up to switch the keycaps, might as well do it.

This is all very helpful. Thank you.

carlca, 2025-05-27:
> [@carlca](https://github.com/carlca), you need a micro USB cable. That was never used on Iphones, maybe you have relatives with an Android phone? Or devices that are a couple years old (e-readers, mp3 players, and so on).

Good thinking! By chance, I had an old Android Phone adapter, and it works. Now I have both lights showing, so I will have to attack my Minichord when my X-Acto knife arrives tomorrow 😉

BenjaminPoilve, 2025-05-28:
Sorry for the scary experience, and glad that you made it. Indeed the Teensy is designed to allow for this cutting so there is no other close trace, making it quite hard to mess up (and I have to thank Paul from PJRC for this great design)!

BenjaminPoilve, 2025-05-28:
@linzqqq, thank you very much for taking the time to report your experience! I'm glad this did not spoil your appreciation for the minichord :)

jimmyslo, 2025-05-28:
Writing from Ontario, Canada. we just received our Minichord today. It arrived much faster than I expected. Your instructions and the video were extremely helpful. As well as the tips from the other folks in this discussion. The right size screwdriver is essential. I was having a bit of difficulty until I found the right one. It felt like I was going to strip the screws. But after that, everything went very smooth. I just took my time.

This instrument sounds amazing. This is going to be such great fun. I bought this for my son who has a disability and is unable to play a musical instrument, which has always been a bit sad for him because he loves music so much. And now with this beautiful little instrument he can experience the joy of playing music on his own. He's loving it. It brings me such joy to see his excitement and the smile on his face. Thank you so much, Pierre!

BenjaminPoilve, 2025-05-29:
@jimmyslo, that is very touching to read. Thanks for the message and happy it can bring joy to your son!

synthmod1, 2025-05-29:
Thanks to all those who have updated with their experience of the modifications-mines on its way and I’m having major FOMO hearing the positive comments about Minichord

microbians, 2025-05-29:
That cut work with my unit! :) Thanks!

soulpixel, 2025-05-30:
Performed both procedures flawlessly, clear instructions many thanks!
Also you can cut the edge of the usb cable, it’s like a plastic lip that edges the micro usb head,so that it fits into the hole of the minichord.

My only question is that I find the overal output of the unit through the headphone cable, the sound level if you want is rather low, is there a way, via the software or otherwise to increase it?

BenjaminPoilve, 2025-06-02:
@soulpixel: to keep the discussion separated, maybe open a different issue? This is a interesting question (and really boils down to the headroom needed for so many independent voices), but would be better discussed in another thread :)

supergranular, 2025-06-02:
I've managed to remove one screw using a 00 Phillips head. The rest seem to have been tightened by a proverbial gorilla. Worried I'm going to ruin it, but can't see any other way other than brute force. Anyone else having this problem?

carlca, 2025-06-02:
> I've managed to remove one screw using a 00 Phillips head. The rest seem to have been tightened by a proverbial gorilla. Worried I'm going to ruin it, but can't see any other way other than brute force. Anyone else having this problem?

Have you tried with a PH0 head rather than a PH00 head? A PH0 worked perfectly for me, and at no point did I feel that there was any chance of shearing the screws.

BenjaminPoilve, 2025-06-03:
@supergranular sorry that you are having this issue. Did we discuss by email? I mentioned to somebody that maybe as a last resort, trying to make the screw colder by using compressed air can might help to unstuck the screw. In any case if there is too much difficulty, Seeed will propose a replacement. 

@samjhill, I opened a thread to discuss alternative keycaps! It is here : https://github.com/BenjaminPoilve/minichord/issues/24

drfunkmeister, 2025-06-03:
My minichord arrived today!  Both fixes have been successfully done - thanks for the clear instructions.  I had no solder blob but it still took three attempts with the X-acto to cut the tiny USB bridge.  If anyone else is doing this and it doesn't work first time, don't worry, keep cutting!  And make sure the cutting blade is sharp.

Great work @BenjaminPoilve , I'm loving it!

panwiewiorka, 2025-06-08:
Hi! Here's yet another feedback:
- The skrews were really tight, had to try ph00 & ph0 but eventually ph1 helped. But there was a moment I thought I'd give up.
- The bridge cut fix was ok, although I've had to cut deeper than it looked like to finally cut it (that's an advice for other users - try more if it looks like already cut but with no effect)
- The main LED is rather bright, it would be great to either reduce it's brightness or omit the cutout and hide the LED underneath the semi-transparent panel surface, giving it nice matte look.
- Can you link to a file where the Preset -> LED color mapping takes place? I want to try to change colors of some presets to distinguish them better. 

Everything else seems great!
Thank you!

electricflock, 2025-06-11:
Did the fixes to my unit. The trace cut was easy enough, buttons just a little fiddly. 
But a few of my screws are not going tight so the faceplate rattles now. I was very careful on the removal and reassembly but some plastic dust came out with the screws so I fear that is as tight as it will go now.

Fizz-Wizardly, 2025-06-24:
whoops, I just saw this thread and have been playing around with this thing for several weeks now, I thought something was off! lol my unit was affected but both of the fixes went smoothly, and my unit is in working order now. 

a potential improvement could be using allen-type socket head cap screws and threaded inserts for the screws that fasten the two halves of the shell together. not sure if you would need to make the shell thicker to accommodate threaded inserts, but just an idea to potentially add longevity to the fastening ability of the screws over time.

I have been absolutely loving this thing and I am impressed with all the hard work put into it. Thank You Benjamin Poilve for making this amazing instrument and for making it available to everyone! It's been an absolute joy to come across this project.

BenjaminPoilve, 2025-06-25:
@Fizz-Wizardly thanks for the suggestion! Indeed I contemplated using threaded insert, but it added significant width to the enclosure walls, so it was not really practical. In additions, in my tests I never had an issue with degradation of the thread in the enclosure, I have some enclosures that I have opened and closed tenth of times. I think the main issue here is the quality of the screws, plus the fact that Philips trace makes it easy to use a wrong-sized bit. Your comments made me check, and it seems that there is some M2.3 screws that have a Hex trace. I'll order some to test them out, maybe that'll alleviate the issues that some users have been facing. 

Happy that you still like it and enjoy making music music with it :)

synthmod1, 2025-07-23:
Me too😂 my replacement arrived yesterday and charging lights and correct notes all good. Had a quick play with the editor yesterday and the random button that Benjamin implemented does throw up some great sounds and shows the depth of the synth.
I’m looking forward to falling in love again with this very special little machine ❤️