BenjaminPoilve/minichord #20: Dim LED brightness

arnoson, 2025-05-30:
Just received my Minichord from seed studio and I love it! Thanks for this awesome instrument :)

When playing it in a darkened room to LEDs are quite bright. Are they PWM-able and could be dimmed thru software, maybe even as an option in the minicontrol?

arnoson, 2025-06-06:
Thanks for looking into this so quickly! I just tested it and seems that this setting is per preset rather than globally for the device. For me it would make more sense to store this outside the preset since this is a general setting on the device.

BenjaminPoilve, 2025-06-06:
That's the one use case where I would agree with you. Unfortunately, the minichord is designed with the idea that all user-settable parameters go through the preset model. There is no "global parameter" system. 
For now I'll leave it be that way, if any further need for global parameters arise, I'll think about adding some control there too.

arnoson, 2025-06-06:
Alright, thanks!