BenjaminPoilve/minichord #21: Feature: change Octave

arnoson, 2025-05-30:
Would be great to be able to change to octave up or down, either of Chord and Harp separately or at least for both of them at the same time. I really like the low sounds of the synth engine. Maybe the preset buttons together wirh sharp button could be used. To indicate the octave we're in the preset LED could pulse faster or slower for a short moment, similar to hove some midi keyboards indicate the octave (I think the Aturia Microlab does this)

arnoson, 2025-06-02:
Thanks for the detailed answer! Didn't know about the setting in minicontrol yet, this is definitely a good workaround. I think it would be still nice to be able to adapt the octaves while playing without having to connect to a computer but I totally get that you wan't to keep the button combos simple. Regarding the preset I think they don't have to sound good / have to be tested on all octaves, I think of it more like an exploration a user could make while playing. A bit like playing a sample really slow on a sampler and hear how the sound reaches its limits. But again, since this is already possible via the minicontrol I can totally understand if you won't add it :)

BenjaminPoilve, 2025-06-02:
My pleasure! Thanks for proposing ways to make the minichord better :) 
Yeah sometimes I think it is a good idea to enforce some limitations. Event the secondary function for the potentiometer can be a bit confusing (I'm still a bit split on whether I should have left that function or not), so I'm really quite reluctant moving forward with the combo functionality. 

I'll post an update this Friday after some testing. I'll try to see how the minicontrol option works out! 
Please note that if you want to move up/down octaves during a single song, it will be possible to save "duplicates" of presets (with different octave selection)  in the minichord memory slots to be able to move up/down presets during a performance.

BenjaminPoilve, 2025-06-06:
Hi, 
The feature is now added. 
To test:
- please update your firmware following [this procedure ](https://minichord.com/user_manual/#updating-the-firmware)
- head to https://minichord.com/minicontrol/ 
- use the parameter "octave change" on chords and harp section

Let me know if that works! 

By the way I had to make a "breaking change". That's not too much of an issue as wiping the presets back to stock solves the situation.

arnoson, 2025-06-06:
I'm trying to test this, but somehow the minicontrol software doesn'r recoginzed the presets anymore. I can connect the minichord but changing the sliders in minicontrol doesnt' change the device and vice versa pressing the preset button up/down on the device doesn't select the preset in minicontrol.

BenjaminPoilve, 2025-06-06:
That's strange. Are you using minicontrol with Chrome as recommanded? With a functioning data cable? 
You updated the firmware? In any case I've kept one with the original firmware and it stays compatible.

arnoson, 2025-06-06:
Never mind, seemed to be some issue with my browser. I used chrome, but for some reasons it crashed a few times, maybe it wasn't handling all the midi. Restarted it and now everything works as expectetd! Thanks for adding this so quickly