BenjaminPoilve/minichord #25: Feature: 'attack transient' 2nd oscillator for harp

drfunkmeister, 2025-06-03:
Would be great if possible.  I'm thinking about things such as experimenting with a short attack transient several octaves up from the fundamental pitch, to create sparkling bell sounds.  So the 2nd oscillator would not need a full set of parameters, and could be limited in purpose to just adding a transient to add some edge or sparkle to the the harp tone.

BenjaminPoilve, 2025-06-04:
It is not impossible per say, but a bit cumbersome to add without making breaking changes. 
The synthesis was designed with a subtractive synthesis mindset. In that context, transient are created by using oscillators with high overtone content and a fast closing low pass filter after the initial attack. Doesn't using that method helps attaining the sound you have in mind?

drfunkmeister, 2025-06-04:
Yes, a fast closing low pass filter definitely helps to give an edge to the sound.  But the method of using a second tone at a much higher pitch is a little different.  The idea is to add a short 'strike tone' and simulate a bell.  For example, a short sine wave transient at a major 7th interval from the fundamental pitch, several octaves higher, can give a nice metallic edge that would be impossible by filtering only.

But if it might break some things, maybe not a good idea!

drfunkmeister, 2025-06-04:
Ha, I'm happy that you're curious about this now... :)  I don't know if it should be added before the low pass filter, or left unfiltered.  But here's a link to the organ service manual, which was very well written, and has all schematics and very good circuit descriptions: 

https://we.tl/t-ae0CHIgHQv

The PDF is text searchable, so you can search for 'strike' and you can see how they did it.  It's an interesting look at the technology of 1980!

BenjaminPoilve, 2025-06-20:
I created a feature there: https://github.com/BenjaminPoilve/minichord/tree/feature-attack_transient 
To test it out, you can upload the firmware into your minichord (the one from the branch [here](https://github.com/BenjaminPoilve/minichord/blob/feature-attack_transient/firmware/firmware.hex)) and use minicontrol to control the new parameter (again, the new version [here](https://github.com/BenjaminPoilve/minichord/tree/feature-attack_transient/firmware/minicontrol) 

I tested it and made this preset: MDswOzUwOzUwOzUxMjs1MTI7NTEyOzA7MDswOzE5Mjs2MDs0Nzs1NjsxNTk7MTAwOzE1NzsxMDA7MDswOzEzODsxOzA7MDswOzA7MDswOzI3Ozc0OzE7MDswOzA7MDswOzA7MDswOzA7NDsxNDsxMTszOzI5OzE4OzY1OzExMDI7MzszMDI7MTE7NzA7NTsxODs0OzI2OzY7MTs1OzA7NTE2OzE2OzA7MDswOzc2OzE7MTsxMDA7MTsxOzY4OzE3OzE0OzIyOzIwOzA7MzQwOzE2ODI7NzA7NDg7MDswOzEwMDsxODs1ODswOzA7ODAwOzcwOzQzOzEwMDsxMDA7MDswOzA7MDsyMDA7MDsyOzg7NTsxOzE7MTkyOzExOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDs0OzEwOzExOzEwMDs5OzA7MjAwOzQ7ODsyMDA7MDs1MDszNjs3NTs1MDswOzA7MzsxOzE7ODA7MTg1NTsyOzc2OTsxNTsxMTQ7MTk7ODszMjs4MDsxOzE7MDszMDswOzExOzA7NDY3OzA7MzU7MDsyNDswOzM7MTsxOzE7MTAwOzE7MTsxMDA7MTsxOzE7MTswOzM0MTsyMTY0OzcwOzU0OzA7MDsxMDA7MzY7NDI7MDsxOzgwOzE2OzQ7MTAwOzIwOzIxMDE7NzA7MTAwOzY7MjI7NTU7MjswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDsxOzI7NDs4OzE3OzA7MTI7MDsxOzI7NDs4OzE2OzY7MDsxOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7

Let me know if that sounds like you imagined!

terminalwaltz, 2025-06-20:
Not the original poster, but can confirm it works and sounds pretty great!

drfunkmeister, 2025-06-21:
Thank you!!  I've updated the firmware but I don't know how to load your new preset or run the new version of minicontrol. I've tried to run some of the files but they just open as code in my browser.  I've tried to download them and the same thing happens.  Do I have to compile it before it will run?

It's probably something simple but I'm new to the world of Github!

terminalwaltz, 2025-06-21:
Hi! I've actually just gone through this process myself, so maybe I can help.

You'll have to start a small server on your computer and host the new minicontrol index.html file on "localhost:8000". This is the only way to get the MIDI permissions you need to talk to the minichord. Make sure you download the entire minichord directory (or use git clone). Change directory to the minichord/firmware/minicontrol directory, and run this line of python code:

`python3 -m http.server 8000`

You should then be able to open Chrome web browser, go to "localhost:8000" and you should get the new minicontrol page and a prompt to allow MIDI permissions.

I'm on a Linux machine, but I'm assuming this will be a relatively similar process for any machine. Let me know!

terminalwaltz, 2025-06-21:
And once you've got the minicontrol page running, to load the preset -- click "load settings" and paste that long string Benjamin posted in the prompt window. That should do it!

drfunkmeister, 2025-06-21:
Thanks for your help @terminalwaltz !  I'm much closer now - the key to it was downloading the entire minichord directory.  Now I can run minicontrol in Chrome (Win10) without needing to do anything else.  I have MIDI access to the minichord and I pasted Benjamin's preset but the attack transient features are greyed out in minicontrol and I can't hear any difference if I tweak them.

Did this happen to you?  Maybe I didn't flash the firmware correctly?

terminalwaltz, 2025-06-21:
Excellent, you're definitely close. Sounds like you need to reflash the firmware -- make sure you're grabbing the firmware from the [branch that contains the attack-transient stuff](https://github.com/BenjaminPoilve/minichord/tree/feature-attack_transient/firmware). Report back if need be!

drfunkmeister, 2025-06-21:
@BenjaminPoilve great work, thanks!!  Yes, it does sound like I imagined, especially with the harp in chromatic mode.  Great result!  I think the range of controls is perfect, and I can only think of one thing that might improve it - to allow the option for the transient to be made louder.  Maybe to make an extra 2 or 3 dB of gain available would be enough?