BenjaminPoilve/minichord #27: A place to share patches

hapiel, 2025-06-04:
Perhaps there could be an 'official' place to share patches. Some suggestions:

- A discord server
- A subreddit
- patchstorage.com
- This issue thread

I would love to hear what you've made the Minichord sound like

hapiel, 2025-06-05:
> Oh that's a very good proposal. I have an email address setup ([presets@minichord.com](mailto:presets@minichord.com)) but receive no submission until now!

I've sent an email to this address just now, but it could not be received
```
<[presets@minichord.com](mailto:presets@minichord.com)>: Recipient address rejected:
       undeliverable address: Mailbox might be disabled, full, or may
       not exist on the server. Reason: JFE030050
```

BenjaminPoilve, 2025-06-06:
> I've sent an email to this address just now, but it could not be received

Thanks for checking. That is solved now :) 

From Seeed's information, they shipped about 300 this month, so maybe I'll see an inrush. But a working inbox is a prerequisite.

BenjaminPoilve, 2025-06-13:
Here is a first test : https://minichord.com/minicontrol/sharing.html 
Let me know what you think!

BenjaminPoilve, 2025-06-14:
Those are all excellent feedback! I'll take that into account for the next version!

BenjaminPoilve, 2025-06-15:
Let me know what you think: https://minichord.com/minicontrol/minishop.html (I renamed the page). 
Regarding the storing of the name, it is not impossible but would have to be limited in size and character to not take too much space in the preset memory. For now this is not my highest priority, but I'll keep it in mind ! 
By the way since the minishop is a fully static website, it would be easy to have a "private library" on your computer, just by having a personal JSON files storing your presets.

YonkRaccoon, 2025-06-16:
Fully static website is a godsend! I can't wait to make my own local version of it with my own presets, and also submit my creations to minishop (cute af name idea and color-coded site.)

BenjaminPoilve, 2025-07-03:
Closing since the solution seems adequate!

closed 2025-07-03 by BenjaminPoilve