BenjaminPoilve/minichord #28: Feature request: Better control of parameters on minicontrol

hapiel, 2025-06-05:
It's hard to slide the minicontrol sliders to a specific value. This is important for many settings, for example the frequency multipliers, I find it incredibly difficult to set them to exactly 1.00

I propose 2 solutions:

- Make the sliders wider, like at least 3 times or so
- Put the values in an editable textbox, where values can be typed. Typing a new value should also update the slider.

BenjaminPoilve, 2025-06-06:
Thanks! The whole interface is quite rough indeed. I take note and will check it out, but frontend is not my forte :)

BenjaminPoilve, 2025-06-08:
By the way: did you notice you can use the direction keys of the keyboard to do some precise settings?

hapiel, 2025-06-08:
Ah no, but it works great, would have saved me a lot of zooming in and out! I suppose that could be documented on the minicontrol page.

It's a bit funny on the logaritmic scales, such as 'release', if I'm at `5000` a single keypress takes me to `4991`, but when I'm at `1` it takes 236 keypresses to get to `2`. Perhaps show a few decimals?

BenjaminPoilve, 2025-06-08:
yes I definitely should! 
I get your point on the logarithmic scale, the issue being that I'm mapping the underlying values that are linear to make it easier to user, and the data is send as an integer (if not, it is displayed with two decimals). So indeed at the lowest range you might have multiple arrow press to jump to the next integer. There might be a better way to do that, but I did not find a great example.