BenjaminPoilve/minichord #31: Harp notes stick sometimes

hapiel, 2025-06-06:
Sometimes the notes on the harp remain sustained when I remove my finger. It has happened more today than on other days, I think the humidity is a bit higher today than usual.

I suppose there is not an easy way to adjust the sensitivity of the capacitive sensor?

BenjaminPoilve, 2025-06-07:
I’ve run into this once. I never could find a fault with the capacitive sensor, signals were always sent correctly. What I suspect is that there is an underlying bug in the audio synthesis library, so this bug only happens on specific presets. If you could send me the DIY (I presume) presets were it happens, it would help me track down the faulty component and make a PR to the audio library.

hapiel, 2025-06-08:
Indeed, it happens when I use this preset:

`MDswOzUwOzUwOzUxMjs1MTI7NTEyOzA7MDswOzE5MTsxMDA7MTkwOzQwOzg4OzEwMDsxNzY7MTAwOzA7MDsxOTA7MDswOzA7MDswOzA7MDswOzUwOzA7MDswOzA7MDswOzA7MDswOzA7MDsxMjsyOzE7MzsyNTk7NTQ7NjA4OzI4OzIwMDA7MDs5MjsxNDMwOzE7MTsxMDA7MTsxOzc1OzA7NjA0OzA7MDswOzA7MTsxOzE7MDsxOzE7MDsxOzE7MTsxOzA7MDswOzcwOzA7MDswOzEwMDswOzA7MDswOzk0MjsxNTY7NTA7MTAwOzEwMDswOzcwMzs2NTszODk7NjM7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDsxOzIwOzY7MTAwOzEzOzA7NTA7MjA7MzsxMDE7MDs2NTs1OTs1NTs0MDs0MzswOzEyOzY7OTk7NjU7MzQ7MTszODg4OzA7NzA7NTAwMDsxOzE3Njk7MzM7MTsxOzA7MTAxOzYwOzY5OzA7MjY7MDs2OzA7ODI4OzA7MjY7MTsxOzE7MTAwOzE7MTswOzE7MTsxOzE7MTg7MzE2OzEyNTY7NzA7NjA7MDswOzEwMDs1ODsyNTsxODswOzE2OTsxNjsxOzEyMjsyMDU7MDs5NzswOzEwMDsxMDA7NTA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDs5OzA7MTI7MTY7MTswOzI7MTI7MTswOzEyOzA7MTsxNjszMjs2NDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOw==`

BenjaminPoilve, 2025-06-08:
Thanks, I can confirm. 
Not clear to me why it causes that issue though. I thought maybe too short an envelope could be the reason, but the envelope is not out of the ordinary.
So currently no clear idea on what triggers the bug but I'll keep thinking about it.

BenjaminPoilve, 2025-06-08:
Found the issue! If the retrigger release is too long, and there is another noteOff while it is waiting for the note to release before properly triggering the noteOn, it never turns off. This is indeed in the underlying audio library. 
Somehow fixed by limiting the range in the front-end, but would need to report the bug to Paul.