BenjaminPoilve/minichord #35: Feature unlocked! "Barry Harris mode" -- How to contribute new features?

terminalwaltz, 2025-06-15:
Really enjoying the minichord, and I've finally got my head wrapped around your programming workflow. I was able to create a feature I call "barry harris mode", which changes all of the major chords to major 6, the minor chords to minor 6, and the diminished chords to fully diminished. I can go on at length about why this would be useful, but suffice to say, it creates a completely new palette of tonal color.

I'm a bit new to Github, but I'm wondering if this is something I could add to a fork of the project? Or is there another way you'd like to handle it? I really think users would get some good use from this feature, it's great for jazz especially.

I've also seen numerous housekeeping bits I could clean up in the code, just spelling and some English syntactical stuff. And maybe some theory stuff, like using "scale tones" in some places for clarification. Is this something I would handle in a fork, and then make a pull request?

Thanks for entertaining some silly questions while I figure things out :)

[image]

BenjaminPoilve, 2025-06-16:
Great to know you managed to get the architecture! 

I'd really love for inputs and error correction to the code. As I said before, I'm sure there are tons of stuff that could be corrected or made better, as well as new features like the one you propose. Indeed standard workflow is to fork the repository, make the modification then create a new pull request from the Github interface referencing the fork in question. You have some useful documentation [here](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork). I'll review it and integrate it in the main branch. 

Regarding your new feature, I can't wait to test it out! I'm wondering: do you think the proposed name is the most explanatory for this feature? I'm wondering if we could find something more explicit for musician (but maybe this is my lack of familiarity with this musician that shows).

terminalwaltz, 2025-06-16:
Submitted a pull request, let me know what you think! 

I'm regards to the name, I'm sure there's something better we could consider. Barry Harris had a specific theory on harmony that has a lot of interesting applications in jazz, especially. He wasn't very well known as a performer, I believe, but his legacy as a teacher remains strong. 

If you'd like me to do a small write-up about it, I'd be glad to. Users might like to know how this option could contribute to their harmonic choices. For instance, a C major 6 is comprised of C E G A, which is the same spelling for an A minor 7. On the minichord, this means you'd have two different inversions of C6/Am7 available, and each of them resolve perfectly from the same fully diminished chord (in this case D,B,F, or Ab diminished, all have the same chord spelling). There are only 3 fully diminished chords due to their symmetry, so these are also represented in multiple inversions. 

So, in terms of fingerings, with the Barry Harris option, you could play Dm7 with the bottom two rows, then on the same root play the top two rows for Ddim. This will resolve beautifully to C6, Am7, Cmaj7. This would make a solid ii - V - I type progression , with some nice colors. OR you can pivot on this diminished chord and point at Em7/G6, what Barry Harris called "the 6 on the fifth". Em7 (E G B D) is a reasonable spelling for a rootless Cmaj9, and thus can also act as a tonic harmony.

Anyway, play around with it, and keep in mind the idea that C6 and Am7 are interchangeable. Let me know how it goes!

terminalwaltz, 2025-06-16:
Yes, I'd be more than happy to do a write-up on the feature for the newsletter! And I'll see about contributing a preset to demonstrate, though honestly, I've been using your Twin Peaks Lynch-inspired preset to great effect so far. 

I think for anyone who has loved the Omnichords of the past, the ability to play unusual chord extensions is an incredible feature, something the original simply wasn't capable of. This may be a feature to expand upon in the future, perhaps it could turn into a more generalized "choose your own chords" for the layout eventually.

Glad you're enjoying the feature, I'll be in touch with a write-up!

closed 2025-06-18 by BenjaminPoilve