BenjaminPoilve/minichord #40: Bug -- MIDI OUT for harp (CH2) ceases when volume pot turned down

terminalwaltz, 2025-06-22:
To reproduce: observe MIDI channel 2 function, turn down second volume pot (harp pot), MIDI transmission through channel 2 ceases. Interesting behavior, doesn't happen for chords on channel 1, but only for the harp notes on channel 2.

I'll have a look and see if I can spot anything obvious, just documenting for now.

BenjaminPoilve, 2025-06-26:
Just to report the discussion here: I think you are very right that velocity and volume are different notions. Still, velocity is one of the midi parameters that is the most commonly supported in MIDI synth, so I thought it made sense to map it there. I've seen it being used by minichord players.
On the other hand, I understand that it might be surprising if you don't expect it. So I'm still split!

terminalwaltz, 2025-06-26:
I think we're maybe missing each other's points, and it's probably because I don't fully understand how you've structured the internal Teensy synth. Forgive me trying to gain some clarification again:

My expectation would be that I could turn off the local sound but still transmit MIDI. Usually this is a switch like "local off". I presume that this simply cuts audio output but maintains all MIDI data transmission. For a velocity sensitive board, like a piano controller, this MIDI information would include variable velocity information, ranging anywhere from 1-128.

The minichord does not appear to be able to produce variable velocities, all notes ultimately come out at the same uniform volume, controllable by the volume pot. If I'm understanding your code correctly, you've used velocity as the stand-in for volume in the Teensy because you're communicating to the internal synth with MIDI data and velocity makes sense as a stand-in for volume in that scenario. 

If this is the case, then I understand why you wouldn't want to break anything by choosing against velocity in your internal framework, and this "bug" makes sense. Afterall, my problem is fixed by turning off my external speaker, it's not a major issue. My thought that this could be a bug stems from my prior expectations of MIDI controllers, this project need not follow suit exactly :)