BenjaminPoilve/minichord #46: Discussion - Minicontrol Interface

BenjaminPoilve, 2025-06-25:
I open this thread to discuss the minicontrol interface. As is, the interface is quite rough, and it would probably make it easier for people to get started using it if it was a bit easier to use. In addition, there are several basic usability issues that need to be disabled. 

Please use this thread to list the issues you are facing, as well as to share inspirations and example that would be useful for the rework. It will serve as a reference when the work begin.

BenjaminPoilve, 2025-06-25:
From my side, an initial list:
- The long list of parameter makes it easy to get lost, and edit the wrong parameter (for example editing Harp envelope instead of Chord envelope) 
- Sliders are not the right interface for all parameters. Need to have more diversity in interface type (buttons for example) 
- Need to allow setting a specific value, for example by bringing up a text input 
- Some people are using minicontrol on smartphones. Need to make sure that the interface works well in that context (tooltip for example does not work there).

BenjaminPoilve, 2025-07-11:
Additional feature: the ability to copy only chords or only harp to another preset, per #53!

terminalwaltz, 2025-07-12:
I want to see a "Simple Mode" or a "Kid's Mode" that simplifies the number of parameters and constrains the potential sounds (to mitigate harsh parameters).

I have an interest in putting the minichord in the hands of my students, ages range from 5-15. The minichord is an excellent vehicle for teaching music theory, sound design, and coding, so I want to be able to easily access all of these features and have students interact meaningfully no matter their age.

Some of my students are more into the tech side of music than theory or performance, they would be thrilled to be able to modify their own synthesizer 🎹 🎶

BenjaminPoilve, 2025-07-13:
Great suggestion! I think some functions are easy to "simplify" (the effects mostly), but there would need to be some thinking about some that are quite cumbersome (the envelopes for example). Maybe we could do an "envelope generator" based on two parameters (strike, decay for example)? I'll give it some thoughts!

terminalwaltz, 2025-07-13:
Want to elaborate slightly on a "collaborative synth patch maker". I don't know if I mean something like "collaborative live editing" as much as I mean maybe a minishop-style interface that just allows users to submit modifications easily, but with some nod towards not destroying the original preset and getting some kind of "approval" from users, maybe? It's a moonshot idea, just figured I'd get it out there!

vincent-maurer, 2025-07-13:
Minichord is a great instrument to play with! I feel that the interface could have some more visual feedback and clustered groups of controls. 

Maybe the different sections could be grouped together with previews for the waveforms and envelopes.
I cobbled together a non functional mockup with the help of AI. It's missing a lot of parameters, but it is  going in an interesting direction:
https://codepen.io/Vincent-Maurer/pen/VYvYqmV

Ideally a visualizer would make the tooltips less important and make it a bit friendlier for beginners

terminalwaltz, 2025-07-13:
@vincent-maurer I really like what you've done here, you should check out [the monstrosity](https://terminalwaltz.github.io/minicontrol/) I've been building. It's got a [repo](https://github.com/terminalwaltz/minicontrol), you should pop over and see if there's anything you can add. I'm largely using AI to help, as well, and as Benjamin as warned me, it has led to a bit of tech debt in the code. But overall it's functioning well, but *use with caution* as it may corrupt your parameters and should be considered "experimental".