BenjaminPoilve/minichord #49: Info on ordering Keymaster Touchzone

Fizz-Wizardly, 2025-06-25:
I am interested in getting one of those "Keymaster touch zones" made but have never ordered a custom PCB before. 
not sure if it's too big of a question but do all of the files in the folder included in (hardware/PCB/touch zones/keymaster PCB/gerber) need to be given to a custom PCB website or just specific ones? 

Does one need to download KiCad and assemble the gerber files and export a completed assembly file of some sort to order?
Or are the files provided in the GitHub folder sufficient for ordering the Keymaster touch zone? And are there any specific instructions to be sure to include in the order when ordering one/them?

Thanks in advance from my fat fingers

terminalwaltz, 2025-06-26:
@Fizz-Wizardly For whatever it's worth, I'm actually waiting on a batch of keymaster touch zones right now! It's a 5 PCB minimum order, so I'm going to have extra, and I'm planning on sending one to another user when they arrive (I'll even solder the headers). If you want one, I'm happy to send an extra!

But otherwise, if you want your own batch, you should just be able to upload an archive of all of the PCB files for the keymaster, a .zip is usually what it calls for. If you're using JLCPCB, you might even be able to use my order number to just get a repeat of the same thing I did: order# Y1-3681988A

Happy to help if you need it! Otherwise, if you want one from me, hang tight and I'll update when they arrive. They shipped out a few days ago, hopefully it won't be long!

Fizz-Wizardly, 2025-06-26:
@terminalwaltz thanks a ton for the info! I’d be interested in one of your extras! I’d be happy to Venmo you some cash to cover any expenses plus some for going through the effort. It would be very much appreciated! 
Hopefully your response will be helpful to some others interested in ordering these also.

Can I Get an email address from you so I can send you my shipping info and get your Venmo or something? 

My Email is jdneff1@gmail.com just send whatever message when you’re ready and I’ll give you my required info. 

Thanks again, YOU ROCK Terminalwaltz!