BenjaminPoilve/minichord #50: Feature Request: Change octaves for set of chords based on the root note

jcirimel, 2025-06-29:
Often times I'm playing and need to reach a chord in a lower octave so I've been using two identical banks with different octaves and switching between them which works but is clumsy. However, rarely do I need to use the same chord in two different octaves. Presently C is always the lowest chord and B is always the highest. It would be very continent to be able to set the effective lowest note from C to another note. This can kind of be done now with the transpose feature, but it would reorganize the notes on the minichord which is undesirable. Ideally I think this could be implemented the same way as transpose with a 12 option selector. Let me know what you think and thanks for your consideration!

terminalwaltz, 2025-06-29:
If I can try to put it in different words for my own clarity, you're effectively describing a "frame shift" of the chords so that the lowest-pitched chords aren't necessarily the chords to the left all the time, is that right? Makes sense, would be a good feature.

jcirimel, 2025-06-30:
Yeah, I think frame shift is a good way to put it. As an example, a frame shift of 0 could have the chord buttons be F4 C4 G4 D4 A4 E4 B4 and a frame shift of +3 could be F4 C5 G4 D5 A4 E4 B4. The frame is shifted to the right 3 notes so C, Db, and D get pitched an octave up.

BenjaminPoilve, 2025-06-30:
I had trouble getting my head around it, but the term frame shift indeed helped to conceptualise the feature. 
I think we need to talk about the behaviour of the sharps. Currently they solely add (or subtract in flat mode) to the current played note. So the basic way to implement this would to have not 12 but 7 level and frame shift only on notes from the C scale. 
If you really see a need for a semitone-level frame shift, that would be slightly trickier to implement, plus I'm worried it would break the mental model of that button (a simple "bend" one semitone). 
Let me know what you think!

BenjaminPoilve, 2025-06-30:
I think the feature request is now quite clear.
From a practical perspective, it'll be much easier to implement a 7 level version. I would keep a map of the seven option, with the associated octave shift. 
Currently the root is stored like that 
```
uint8_t root_button[7] = {11, 4, 9, 2, 7, 0, 5}; // defines the fundamental of each row in the circle of fifth, ie F,C,G,D,A,E,B from left to right
```
It could be changed to something like that 
```
int8_t root_button[7][7] = {
    {11, 4, 9, 2, 7, 0, 5};  // C as lowest
    {-1, -8, -3, -10, -5, 0, -7},  // D as lowest
    {-1, -8, -3, 2, -5, 0, -7},  // E as lowest
    .....,  // and so on 
}
```
I'm not sure how clear it would be to a user that did not follow that discussion though. Both the feature and the implementation.

terminalwaltz, 2025-07-10:
@jcirimel Give this [new firmware.hex](https://github.com/terminalwaltz/minichord/tree/feature-FrameShiftChords/firmware) a shot and let me know if this is what you were thinking.

You'll have to run this [modified minicontrol](https://github.com/terminalwaltz/minichord/tree/feature-FrameShiftChords/firmware/minicontrol), too, so you can actually change the values for "chord frame shift", let me know if you need help!