BenjaminPoilve/minichord #53: Feature Request: Copy only chords or only harp to another preset

terminalwaltz, 2025-07-05:
John W dropped some excellent synth patches in the minishop and inspired me to spend some time crafting my own. I've wanted to pull in a few different sounds from different patches to combine with my own and tweak a few things, would be really great to have the option to import just harp or just chords from a preset.

I think you'd define the indices associated with harp parameters, collect the desired parameters, and splice() desired parameters into the array, maybe two separate functions for harp and chords?

BenjaminPoilve, 2025-07-08:
That's definitely something that would be useful ! I thing we should add that to the list of interface function we would like to add!

BenjaminPoilve, 2025-07-11:
I'll close it because this probably belongs in the discussion about interfaces! Still accessible from there.

closed 2025-07-11 by BenjaminPoilve