BenjaminPoilve/minichord #60: Try this Random Preset Generator

terminalwaltz, 2025-07-12:
[This python script](https://github.com/terminalwaltz/minichord-random-generator/tree/main) will generate (and copy to your clipboard) a preset compatible with the minichord.

It's constrained to choose values and adjust variables that *should* result in musically-reasonable presets, but it's not perfect. I'll probably play with it little bits here and there, but it's too fun not to share.

For a fun exercise, generate a random preset and then try to shape it where you think it was meant to go.

I just shaped this one from a random start, pretty fun. Make sure to twist some knobs, I set potentiometer variables intentionally.

`MDswOzY3Ozc0OzUxMjs1MTI7NTEyOzA7MDswOzYzOzEwMDs5OTsxMDA7ODU7MTAwOzE4NDsxMDA7NTswOzI7MTsxOzI7ODY7MDs1Njs4NzszNzs2MDs0OzE7NTc7MDsxOzA7MDswOzA7MDs0OzM5OzE7MDs1MDM7MjcyODs1OTsxNTE7MzszMTE7MDs1MDA7MTQ3Nzs2ODQ7MjUzOzEwMDs1MDAwOzU4OzE4Mzs2OzA7MTU7NDsyMDAwOzMyOzMwNDg7MzI0ODsyNzQ4OzEwMDsxMTk5OzEwMDswOzA7MDswOzUwMDA7MTAwOzMxNzsyMTQ5OzcwOzM1OzQzOzM3OzI2OzUwOzc2OzIyOzE7NTAwMDsyNDQ7MTAwOzU0OzQ3OzQ7MDswOzk0OzI4OzA7Mjs0OzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MTsyMjs2Ozk5OzIxOzA7MTk4OzE2OzM7MTAwOzA7Mzk7NTU7NDA7Mzc7ODk7Njc7MDs2ODg7Njg4OzUxOzEwNTI7Mzg7Mzc4OzI7MTsxMjI3OzM5OTszNzQyOzMxOzExNTc7NTY7MDsyNzE7MTM7Mjg4OzU7MTg7MDs4OzQ7NjMxOzE7NDk7MTM3NzsxOzgxOzE1Ozk0OTs5OTsxMDM7Mjc7MjA7MjA7MDszOTs0ODg7MjUzNDs3MDswOzUwOzA7OTM7ODg7NzU7MDsxOzI0MjsxMTs0OzE7NjUzOzI5MjM7Mjs0MDs2NTs3MjsxMDE7MjswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDs2MTs2Mjs1OzEyOzM4OzExOzU5OzE5OzU1OzYwOzYxOzI7NTg7NTE7Mzg7NDA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDswOzA7MDsw`

BenjaminPoilve, 2025-07-13:
That was a quite old request to add that feature to minicontrol :)

Indeed this is very fun! The rhythm mode on the preset you shared is really great. Quite addictive too, you just want to see what comes next :) 

I think python is a good way to develop this, and will be quite easy to convert to JS once this is stable. 

Currently I feel there might be a need to "tame" some of the randomness by allocating for a weirdness budget so that if the value is very far from normal on one parameter, it is more conservative on others. Just an idea! 

Even in the current state, it would make a great addition to minicontrol!

synthmod1, 2025-07-13:
Hi Benjamin-that’s amazing .i am still awaiting my return from seeed so hopefully will get mine soon to try this great feature and thanks for listening 👍

BenjaminPoilve, 2025-07-13:
@terminalwaltz, there is an issue with some parameters scaling. I'll look into that, this is very probably my conversion in JS that go defects.

terminalwaltz, 2025-07-13:
@BenjaminPoilve Man, excellent implementation, really smart just to add it directly to minicontrol, thank you for picking this up! Feel free to edit anything I provided to better suit the purpose, obviously :)

BenjaminPoilve, 2025-07-14:
To allow for maintainability, I modified the preset generator to create a Gaussian distribution around the default values, with a "weirdness factor" for the sigma of the distribution. 
It should be more flexible than the rule based version. Let me know if you still like it :)

terminalwaltz, 2025-07-14:
Excellent! I was definitely hoping for a "weirdness factor" that could spit out something more interesting, this is definitely going that direction, really cool!