BenjaminPoilve/minichord #62: Harp Voice Leading

terminalwaltz, 2025-07-14:
I'm already tooling around with this in the firmware, but I would love some input.

In short, I want the harp to maintain a memory of the last chord played, and if the new chord has common tones with the last chord, I want them to stay where they are on the harp for the new chord. I want "smooth voice leading", for lack of a better term.

Playing a song, say something like F#m D A E 😆, I would like my fingers to be able to stay on the same two harp notes instead of jumping just nearby to grab the same notes I was just playing. 

I've been calling the feature "harp voice leading", and it could easily be turned on and off, but I'm having trouble getting the logic right, I think probably because of the interaction with harp shuffle arrays? Just interested in your thoughts on this, I think it would be a great feature :)

BenjaminPoilve, 2025-07-14:
Voice leading has been in my mind for a while, as it could be something that really sets apart the minichord! 

Trouble is, I always have problems defining what should be done: 
- First there is the question of the "distance metric". My first reflex would be to gravitate towards something akin to an earthmover distance: sum the moves in semitone for each voice and try to minimise that. But is it always the case? Do we want to favour zero change for some notes over the general distance (ie find the common notes, then fill the rest)? 
- Then there is the question of the set of chords we want to attain. Is it only the inversions? Do we need to keep the note "ascending" or should it be ok to have them out of order?
- Then there is a musical question: do we always want to have the minimum distance? Voice leading is also about choosing to create breaks, not only to smoothly go from one to the next. 

If we can define the feature properly, it will be much easier for me to identify where it should be added!

terminalwaltz, 2025-07-14:
Excellent, you've obviously given this a lot of thought! Let me see if I can bring perspective from the music theory realm and we can figure out what this thing should be:

In my eyes, "good voice leading" in the theoretical sense would be most easily portrayed by a vocal quartet -- each singer is partaking in a part of the harmony (chord), and [best practice/ease of singing/attractiveness to the ear] means that you want the singers to move their voices as little as possible to achieve the next harmony. 

The application of this concept to the minichord/harp mean we're considering that we have 12 voices (for 12 strings) instead of only 4 voices like a quartet, and we want the availability of those voices to move as little as possible from chord to chord.

Of course, the difficulty with having a pre-established harp layout is the lack of "freedom" to choose certain inversions over others -- as you mentioned, we don't always want the minimum distance, that is the whole point of musical choice, especially in improvisation!

Come to think of it -- perhaps this could be something we think about from the perspective of the chords, as well. Right now, all of the chords are stacked root-third-fifth-octave (with some exceptions for extended chords), but the ultimate freedom would lie in somehow being able to choose from "root" "first" or "second" inversions (which stack the tones in different orders) before pressing the next chord button. This is probably impractical from a performance perspective on the minichord, but I wonder if there couldn't be some internal logic that might calculate which inversion would provide the cleanest movement from the previous chord.

The harp layout is dependent upon chord choice, so perhaps tackling voice leading on the chord level would make more sense??

terminalwaltz, 2025-07-14:
I have to mention this possibility, as well -- now that we have a function to effect a key signature upon the chord buttons, we might have room now to *add a harp array which is simply the major scale of the key signature*. This could potentially simplify the idea of voice leading in the sense that it would be a remapping of the *major scale* every time instead of a remapping of a chord, or perhaps some combination thereof (like prioritizing common tone conservation)? I know you already have harp arrays which add alternate scale tones, but I guess this would be a different treatment of the harp layout based on the entirety of the major scale instead of one chord. I suppose this would break ascending/descending order in some places, but I think some of these tradeoffs would be worthwhile.

I may want to add a harp array for "major scale" based on the key signature now that we've done it, so be on the lookout! :D

keyandcableco, 2026-09-23:
Just to follow up here -- I haven't applied it to the harp yet, but [automatic voice leading](#147 ) for the chord buttons is sounding pretty great. In my [custom build of SoundLab](https://keyandcableco.github.io/minichord-soundlab), you can see on the staff how nice and close the voicings are from chord to chord. This sound much closer to how a pianist would voice the chords. If you play through the diatonic chords in C major, for instance (i.e. C Dm Em F G Am Bdim), you'll hear the voicings climb the scale naturally and there's no weird wrap-around back to the tonic C.