BenjaminPoilve/minichord #72: Compatibility with USB host mode compliant devices

BenjaminPoilve, 2025-07-30:
As you know, the minichord only works with devices that support the USB host mode. I've had reports from minichord users regarding its use. 

There are some positive example, in which all seems to work quite well: 

- Norns (by Monome) 
- Organelle (by Critter & Guitar)

But there are also situations in which it seems that the MIDI signals get scrambled for some reason: 

- RK006 (by RetroKits)
- H2MIDIPro (by CME)

I currently don't have either of those devices, so it is a bit complicated to troubleshoot what the issue could be. I might need to purchase one at some point. In the meantime, I'd like to use this thread to discuss such issues and hopefully get some helps from people that have more experience on the matter to try to troubleshoot those issues.

iktomi-sakala, 2025-10-02:
Hey, I'm really interrested by your Minichord. I have a home studio with some stuff, and even if I consider your instrument already cool by itself, if it would work with some of my synth, it would be really fantastic. (mainly my GR-1 and GR-Mega from Tasty Chips)
I have no expectations, but I think it might work, as they are pretty much "open" in term of connectivity. (based on raspberry, direct usb connection with a Microlab or the Scarlett 2i2 work fine by example, they just need to be class compliant).
I think I'll order you one for what it is anyway (when I can)... and then I'll do some tests, and get back to report what I got.

mio-moto, 2025-10-02:
I have a H4MIDI WC at my disposal and a teensy development setup. I can do some debugging, if you point me in the right directions.

cearka, 2026-01-30:
maybe this is a related issue, but I have some issues getting it to consistently connect to my op1f and opxy.

at first I thought I solved it by configuring a bank to have both chord and harp use channel 1, single port mode.  This is what I need to get it to work when it actually connects.

the problem is 9 times out of 10 it connects, and then immediately disconnects.  the OPxy and 1f both register it as MINICHOR so it does see it.  after unplugging and plugging it in a million times it will work fantastically until I turn either device off

cearka, 2026-01-30:
> Ok, after testing some, it seems like the device has to be powered on (throug a plugged in headphone jack) and be connected to a desktop first. Then it can be, while being kept powered on, connected to hostable USB devices like the move or H4MIDI and it works temporarily. There seems to be an issue with the MIDI setup on USB when no desktop host is present.
> 
> If the device powers on and is connected to just an USB host, it goes into that error state
> 

oh wait, I think this did it for me.  if I unpluge everything, plug it the audio cable, then plug in the device to my pc, then unplug it from my pc and plug it into the OP1, it stays connected.  sometimes.  

doesn't appeart to be a consistent fix though.

cearka, 2026-01-30:
ok I think I found a fix.  in the main.cpp I added a bunch of 100 microsecond delays:

+  delayMicroseconds(100);

after all the note functions: 
Line 464  - after usbMIDI.sendNoteOff() in play_single_note()
Line 467  - after usbMIDI.sendNoteOn() in play_single_note()
Line 479  - after usbMIDI.sendNoteOff() in play_note_selected_duration()
Line 482  - after usbMIDI.sendNoteOn() in play_note_selected_duration()
Line 545  - after usbMIDI.sendNoteOff() in set_chord_voice_frequency()
Line 548  - after usbMIDI.sendNoteOn() in set_chord_voice_frequency()
Line 922  - after usbMIDI.sendNoteOff() in handle_harp() (note on branch)
Line 925  - after usbMIDI.sendNoteOn() in handle_harp()
Line 935  - after usbMIDI.sendNoteOff() in handle_harp() (note off branch)
Line 1001 - after usbMIDI.sendNoteOff() in update_harp_notes()
Line 1003 - after usbMIDI.sendNoteOn() in update_harp_notes()
Line 1023 - after usbMIDI.sendNoteOff() in stop_chord_notes()
Line 1040 - after usbMIDI.sendNoteOff() in handle_rhythm_mode()

I think these delays keep it from overwhelming the buffer on the receiving device.  if this delay is not enough for other instruments haveing similar problems, it can probably be increased all the way up to 500ms.  The miniharp now works consistently with my op1-field and op1-xy.

BenjaminPoilve, 2026-01-30:
I'm pretty sure I tested to add some delays because I also suspected that the midi was overwhelming the receiving device but hey, if that works, that's great news, I might have messed it up :)
I'll check today if your fix works for the H2MIDI. If that's the case that would be a great breakthrough. In any case I'll reintegrate that into the main branch if there is no adverse effect, because OP-1 are a popular device.

keyandcableco, 2026-09-16:
Could anyone here with a USB host box give the fix in #139 a try? You can find patched firmware [here](https://github.com/keyandcableco/minichord/blob/test-allFeatures/firmware/firmware.hex) for testing.

keyandcableco, 2026-09-17:
I don't have one of the failing boxes, so I tested what I could from a Linux desktop: what the minichord actually sends when a host reads it the way a box does, and what a few simple receivers would make of it. I haven't found the bug, but it narrows things down, and there's a cheap way to settle the rest.

## What the minichord sends

Stock v9 sends a **514-byte parameter dump** on the chord port about 1.9 s after power-on, and another on **every preset change**, whether or not anything asked for it. That's the `control_command(0, 0)` at the end of `load_config()`. Both dumps arrive complete and well formed (`F0` … `F7`).

Two things I suspected turned out not to happen, at least at high speed:

- **Truncated dumps.** The device is configured about 2 s before the boot dump starts, so any host that connects at a normal speed receives all of it.
- **Loss from a slow reader.** With reads paused for 500 ms partway through the boot dump, the Teensy held the rest and delivered it intact.

## What a receiver makes of it

I replayed the recordings through four deliberately simple receiver models. These are mechanisms, not any real product's firmware.

| Receiver model | Stock v9 | With the dump gated |
| --- | --- | --- |
| Spec-compliant parser | OK | OK |
| Box using running status that isn't reset after sysex | OK in this run* | OK |
| 128-byte sysex buffer that falls back to running status on overflow | **191 phantom notes** at the preset-change dump | OK |
| Both ports merged in USB order | OK | OK |

\* It passed only because a note-off came before the dump and a note-on after it. A note-on on both sides, such as changing preset while holding a chord, would lose the notes that follow.

The phantom notes in the third model are built from the preset data, so the same preset produces the **same wrong notes every time**, on whatever channel was last played. That matches the "same weird chords" reports in this thread. It doesn't prove any particular box works that way.

## The fix, and how to confirm it

[#139](https://github.com/BenjaminPoilve/minichord/pull/139) only sends the dump when a controller has talked to the device. minicontrol still gets it when it connects, and still follows preset changes. It depends on [#138](https://github.com/BenjaminPoilve/minichord/pull/138), which fixes the flag it relies on. With that in place, the recording above shows **no sysex at all**.

The quickest confirmation is someone with a failing box (H2MIDI Pro, RK006, H4MIDI, Move, M8) trying a build with the gating. My combined test build has it, along with the same gating for the key-change and double-tap reports: [`test-allFeatures` firmware.hex](https://github.com/keyandcableco/minichord/blob/test-allFeatures/firmware/firmware.hex). The useful checks are:

1. Power on connected only to the box, then play chords and harp.
2. Change preset, then play again.

If both are clean where stock v9 misbehaves, it's the dump.

Still untested:

- **Full speed (12M)**, which is what most of these boxes use. The same dump takes about 11 USB transfers instead of 2.
- **Bus power** from a pedalboard box.

## The tools

Everything is in [`tools/usb-host-test`](https://github.com/keyandcableco/minichord/tree/test-usbHost/tools/usb-host-test) on a branch of my fork: a `usbmon` cold-boot capture, a decoder for it, a "virtual host box" that reads the MIDI endpoint directly and runs the receiver models, and the three recordings above. The recordings replay with `python3 virtual_box.py --replay captures/ben-v9-vbox.json`, without a minichord. It's Linux only, and none of it touches the firmware.

BenjaminPoilve, 2026-09-18:
Well I'll be damned! You fixed it! 
It now works perfectly on the H2MIDI! 
That's just incredible work!

jacooob, 2026-09-21:
I tested this fix using a RK006 and it works. Thank you!

Patataonk, 2026-09-21:
ive tested with with H2MIDI hooked to OP1 and it works but only the harp is detected

previous firmware only the chord channel worked for me

I've tried single port mode and it also doesnt work
maybe theres something else I missed?

BenjaminPoilve, 2026-09-21:
Are you sure the H2MIDI is properly routing the signals? It can be configured with the HxMIDI TOOLS software, and maybe the channels are not properly mapped?

Patataonk, 2026-09-21:
> Are you sure the H2MIDI is properly routing the signals? It can be configured with the HxMIDI TOOLS software, and maybe the channels are not properly mapped?

yea that was the issue

I can confirm it works 100% 
thanks!

BenjaminPoilve, 2026-09-23:
I'll close this very long standing bug! Infinite thanks to @keyandcableco for finding the origin of the issue. 
The firmware on main now integrates the fix :) 
https://github.com/BenjaminPoilve/minichord/blob/main/firmware/firmware.hex

closed 2026-09-23 by BenjaminPoilve

BenjaminPoilve, 2026-09-29:
I got multiple confirmation that this solves the issue with the ableton move as well !